CR6Community/CR-6-touchscreen #1: Remove the language switcher

Sebazzz, 2020-09-25:
Let's drop support for Chinese for now. 

[ ] Remove the language switcher
[ ] Remove the toggle button

Sebazzz, 2020-09-26:
@Nushio Can you drop the language option from the 47_E_Control.bmp, then at my side I will remove the toggle switch. Just keep the space empty for now - I don't want to mess with moving up the Restore Factory and Info options just yet.

Nushio, 2020-09-26:
Done. I also provided a pre-exported BMP on the images_source called 47_E_Control_tmp.bmp if you want to mess around moving Restore Factory and Info ;-)

Sebazzz, 2020-09-26:
@Nushio Got it - removed the interface stuff in e96e15b. 

In b27458922 I added a note, apparently any changed BMPs must be placed in the DWIN_SOURCE folder. Otherwise they get overwritten.

commit e96f15b9 (2020-09-26): Drop Chinese language support in touch display

Fixes #1
commit bcfed4c9 (2020-12-06): Drop Chinese language support in touch display

Fixes #1
commit 4bf74d8c (2020-12-06): Drop Chinese language support in touch display

Fixes #1
closed 2020-09-26 by Sebazzz