CR6Community/CR-6-touchscreen #4: Updating T5L_UI_DGUS2 for additional functionality

randomgitname, 2020-09-28:
Cr6 screens ship with version 1.4, current version on dwin's site is 3.5 "T5L_UI_DGUS2_V35_20200820.BIN" @ http://dwin.com.cn/home/Index/download_file?download_id=4608.

This update will enable you to use the cfg generator to turn off touch sound feedback or adjust the frequency/loudness.
You can also turn the screen brightness down and go into a even dimmer sleep mode.
Complete change log is in downloaded archive. I do not believe we need to reflash the T5L_OS_ file as it has not been updated since 2019.

If a bad flash occurs a cfg can be setup to format the flash as a last resort trouble shooting step. (untested)

I leave implementation of this info to you as i suppose some of this can be done in firmware now with the source out. But if you have animals that freak out @ beeps i needed this ASAP. 

Great work so far on the firmware!

Sebazzz, 2020-09-28:
The cfg file can actually be configured separately - no need to flash a kernel. If you look at the newest documentation PDF in the doc/vendor folder, the cfg format is defined there. With a hex editor like HxD you can change the bytes at the right offset (ensure the file signature at the start matches the documentation).

I have seen that they offer a new kernel on the site but I don't dare to let someone flash this:

1. We don't know exactly what type of touchscreen this is. It is a T5L1 with DWIN II but is it also compatible with this new kernel? Does it have protection against flashing the wrong OS.
2. The release notes go back only to early 2020 and this touch screen has a version of 2019 on it.
3. If something goes wrong flashing this OS, can we unbrick it?

If someone wants to sacrifice his or her printer: fine - but I cannot ask anyone to do this for me  😉

Met vriendelijke groet,
Sebastiaan Dammann

randomgitname, 2020-09-28:
I guess I was not clear. I've already done it. Successfully. I was just sharing information in a place that seemed logical.

Yes, the cfg file system works on version 1.4 but the features I wanted to take advantage of were only added after. You also don't need to use a hex editor, the dgus2 8.0.1.1 software has a user interface: click profile generation on the main page, load the default cfg file for editing.

1. true we don't know the model number but I found one that looks alot like it. DMG48270C043_03WTR, and yes it works.
I do not know what sort of protections are in place. The process was no different than flashing the screen. I do not know if rollback is possible. I have a feeling it will just flash anything with the correct name scheme.

2. Not sure which notes those are but as far as the T5l_UI_ firmware: 2019.02.02 was initial release. 2019.04.10: 1.4 cr6se version. 2020.08.20: latest version i flashed and am currently using. I was saying the other file in that package T5L_OS_ has not changed since 2019.7.20 so i didn't flash it. 

3. Fair question. My gut says the bootloader is untouched.

Sebazzz, 2020-09-28:
Which exact binary (path and filename) did you flash from the archive? There are multiple bin files from what I saw.

/cc @nickacker

randomgitname, 2020-09-28:
T5L kernel program2020-8-21.zip\T5L kernel program2020-8-21\T5L│╠╨≥\DGUS2\T5L_UI_DGUS2_V35_20200820.BIN

Sebazzz, 2020-10-27:
I've updated the touch screen to `T5L_UI_DGUS2_V36_20201023.BIN` it seems to resolve some issues with font rendering.

Sebazzz, 2020-10-27:
@randomgitname Would you happen to have this original zip? I'd like to check something.

randomgitname, 2020-10-28:
I don't have the whole zip, delete that as the latest version would always be around. I do have T5L_UI_DGUS2_V35_20200730.BIN, T5L_OS_V11_RTC(RX8130).BIN, T5L_OS_DGUS2_V10(no RX8130).BIN and some english translated pdfs.

Sebazzz, 2020-10-28:
Could you upload them here please?

Met vriendelijke groet,
Sebastiaan Dammann

randomgitname, 2020-10-28:
Let me know when you got it so I can delete.

Sebazzz, 2020-10-28:
And now it's here. You can delete it. Thanks! 
[T5L_UI_DGUS2_V35_20200730_2.zip](https://github.com/CR6Community/CR-6-touchscreen/files/5450898/T5L_UI_DGUS2_V35_20200730_2.zip)

Sebazzz, 2020-10-28:
For extui reverted to your version of DWIN OS - the newest DWIN has random issues by which the screen does not want to start up.

randomgitname, 2020-10-28:
Wow, not something one would expect from almost 2 year old firmware. 
But on the plus side you learned it is possible to downgrade.
It seems dwin failed to update the change log for 36. I also believe there is another version of v35 out that added the last 2 entries in the change log.

-Add icon display to support negative coordinates;
-In the 0x11 text display, support for the Y coordinate of the non-gray font dot matrix text to be a negative number.

http://forum.dwin.com.cn/forum.php?mod=viewthread&tid=2544
if you make an account on this forum you might have access to older versions?

Sebazzz, 2020-11-01:
Keeping it on V35 for now - though every version is probably prone to have bugs :P 

At least glad I could downgrade 💯

closed 2020-11-01 by Sebazzz