CR6Community/CR-6-touchscreen #7: M73(Print progress) amd M117 (Message) support for Octoprint

Ranney1, 2020-10-02:
Would be great if we can find a way to make the LCD switch to the 'print' screen while printing with Octoprint.
Most important functions would be M73 support (print progress) and M117 (display message)

Sebazzz, 2020-10-02:
I'm currently re-engineering the UI on top of ExtUI. That should pretty much automatically cause this to work as expected.

Sebazzz, 2020-10-02:
Reading again, it are three things you're asking:

1. Go to the "print screen" when printing through a host
2. Support M73 (print progress)
3. Support M117 (custom messages)

Ranney1, 2020-10-02:
Dear Sebazzz,

Thank you! That is exactly where I was looking for! Thank you again for the really good firmware and all your time!

Ranney1, 2020-10-02:
Please note that the LCD is not closing the ABL screen when it is finished with leveling. So I think that the 'print screen' must be triggered after ABL.

Ranney1, 2020-10-02:
Maybe that this code is also nice to implement?
M355 - Case Light Control
To turn on/off the led on the hotend by gcode/remote?

Ranney1, 2020-10-04:
Update: LED control is actually possible by GCODE. 
M42 P6 S0 ; is led off
M42 P6 S255 ; is led on

Sebazzz, 2020-10-28:
M355 implemented on extui, as well as M117.

Ranney1, 2020-10-31:
Ahh, really great! Looking forward to the moment that it is also working when printing with octoprint!
Thanks again for your great work!

Sebazzz, 2020-11-01:
M73 is now implemented in extui branch

Sebazzz, 2020-11-01:
Ready for next release.

closed 2020-11-01 by Sebazzz