CR6Community/CR-6-touchscreen #9: (FR) Display axis positioning on print screen

hondrus31, 2020-10-26:
### Description

Not sure if this request should be posted here, on marlin git, or both.
I think that percentage bar showed on screen makes difficult to understand the progression of your print.
While X/Y value is more a gimmick, Z value can be useful. For example: you set a "pause at height" to insert a magnet in your print, reading the exact height of the Z axis will let you know how many layers you have to wait.

I made up an example of how the print screen would look like:
[image]
[37_E_printing.zip](https://github.com/CR6Community/CR-6-touchscreen/files/5441307/37_E_printing.zip) - this zip file contains a bmp image, github wouldn't let me post it here as is.

Sebazzz, 2020-10-26:
Actually, I don't even think showing the Z offset all the time is very useful is it? I think the axis might even be shown under the speed/z-offset part perhaps.

hondrus31, 2020-10-26:
Yeah, Z offset is pretty useless, constantly showed like right now.
And I imagine that swapping Z offset value with Z axis value would be easier to code, right?

Sebazzz, 2020-11-15:
We won't show X/Y axis positioning because the overhead and delay of updating it on screen is quite high. However, the Z-axis position is useful, so @grobux is incorporating that in the redesign.

Sebazzz, 2020-11-15:
There is some space below the Z-offset for the Z-axis position, and at the left side below the "speed dial" there is space for the extrusion rate (flow rate).

grobux, 2020-12-29:
Feature added in `extui-dev`, thanks @hondrus31 for the suggestion.

[image]

closed 2020-12-29 by grobux