CR6Community/CR-6-touchscreen #10: [BUG] bad inverted style for toggle buttons

DreadKnight, 2020-10-31:
Toggle buttons such as Fan and LED should have the circle on the right side when actually turned on, that's the standard way.

Sebazzz, 2020-10-31:
Moved this to CR-6 touch screen repository.

---

I agree - feel free to submit a pull request with new bitmaps for the toggle buttons - or @Nushio are you able to take a shot at this?

DreadKnight, 2020-11-04:
Yeah, had a feeling there's an even better repo for this. I've also found `Prepare/Disable stepper`. If any other inverted toggle buttons in the UI, feel free to list.

Sebazzz, 2020-11-04:
They all use the same icon under the hood, so change one icon and you have them all.

DreadKnight, 2020-11-04:
@Sebazzz Understood, good to know :+1:

Sebazzz, 2020-12-24:
Already fixed by @grobux

closed 2020-12-24 by Sebazzz