CR6Community/CR-6-touchscreen #11: Issue when scrolling in the menu during Octoprint

Ranney1, 2020-10-31:
When printing with ocroprint, there are no problems. But when:

1. Opening the control screen, close it
2. Opening the prepare screen close it
3. Opening the file screen and closing it
==> the extruder is clipping and running (uncontrolled) with a high speed. only an M112 will stop this. I can trigger it now for 2 times. I've got the idea that the problem is in the (new) file screen.

Firmware: Marlin Extui
Screen firmware Extui

Sebazzz, 2020-10-31:
This is *while* printing with octoprint?

Ranney1, 2020-10-31:
Yes, when printing with octoprint

Sebazzz, 2020-11-01:
I let @Doridian reproduce it - he couldn't. Neither could I. 
Can you please try again with the latest sources (both touch screen and Marlin)?

Sebazzz, 2020-11-06:
Closing due to inactivity.

closed 2020-11-06 by Sebazzz

Ranney1, 2020-11-06:
Sorry, I think that my update was not send successfully. In the latest St extui version, the menu options are no longer visible so this problem is solved now. Thanks for closing!