CR6Community/CR-6-touchscreen #14: FAN speed on printing screen

Ranney1, 2020-11-15:
Would be great to show the current fan speed on printing screen.
I'm using the aufo fan speed option when bridge is detected. Would be great to monitor this function!

hughesjasonk, 2020-12-24:
> Would be great to show the current fan speed on printing screen.
> I'm using the aufo fan speed option when bridge is detected. Would be great to monitor this function!

I agree. that would be awesome

Sebazzz, 2020-12-29:
@grobux did this already I think

grobux, 2020-12-29:
I added a lot of things. Unfortunately, I forgot about the fan speed... and now I'm again a little short on screen space.

[image]

Sebazzz, 2020-12-29:
I think this will not fall well with some people, but I think we can loose the Z-offset here. You can see it from the tuning screen anyway.

grobux, 2020-12-29:
It's OK for me. I implement this.

Sebazzz, 2020-12-29:
If it really bothers some people we can move the printing speed (feed rate) to the header. Like "Printing     [100% speed]".

grobux, 2020-12-29:
Screen updated.

[image]

closed 2020-12-29 by grobux