CR6Community/CR-6-touchscreen #16: Question/Offer: contribute translation

rmeissn, 2020-11-16:
I'd like to contribute a german translation for the touchscreen menus. Is there something like a guide on how to do this?

Sebazzz, 2020-11-16:
Internationalization support in DGUS DWIN is very cumbersome and not feasible. The background images of each page has the text hardcoded. To translate and have it first-class, you would need to duplicate all the bmps, give it a separate ID, and maintain that mapping in firmware as well. 

So, what I would suggest you do:
1. Fork this repository.
2. Work on the extui branch (this is the branch for all work going forward)
3. In your fork, follow [the steps in the README.md](https://github.com/CR6Community/CR-6-touchscreen#images--screen-images-sources) to change the current bitmaps and translate them. @Nushio has made XCF files as the source of these bitmaps, usable in Gimp, to make life easier but you can follow your own way of course.
4. When we change something, it is up to you to replicate those screen changes. Therefore I recommend only to update the screen backgrounds and don't use the DWIN editor other for the purpose of generating the ICL.
5. German touch screen builds are up to you to provide which would then need to be made from the same moment as we are releasing builds.

Good luck, and if you have any questions feel free to use this issue for it. You can also drop by in our Discord.

closed 2020-11-16 by Sebazzz

rmeissn, 2020-11-17:
FYI: Creality released a new Firmware today (v2.0.1.2), which offers various languages for the touchscreen. They didn't hardcode these to the images, so it must be somewhere within the firmware. Unfortunately they haven't released the sources yet (I think).

Sebazzz, 2020-11-17:
They hardcoded this in the DWIN icon files (ICL) - this would require maintaining a mapping in firmware. This is not feasible, especially not for the few working on this firmware. We would also need to replicate (=reimplement) the changes.

Because of this, it is best if someone wants to have a language, that person maintains his own set of background images and distributes that.