CR6Community/CR-6-touchscreen #17: Can't run DGUS software

Tillema, 2020-11-16:
Hi,

I really really really want to start tinkering with the touchscreen, but unfortunately, Th DGUS V8 software doesn't work on my PC (Windows 10, 64-bit). Does anyone know a solution or other software that works with this?
I want to upgrade the interface to BTT-style since I won the replacement board a few days ago.

Regards,

Simon

grobux, 2020-11-16:
Hello,

I'm also on Windows 10/64 bits, and can run the program.
Do you have an error message on screen?
Which executable do you run?

Sebazzz, 2020-11-16:
"Does not work" is the worst way to report a bug ;)

Don't forget to make your changes on the extui branch and sync with the devs in Discord if you really want to work on something, because DWIN files cannot be merged.

Tillema, 2020-11-16:
Sorry for being so general. I don't have a lot of experience with github.
The error I get is "The code execution cannot proceed because Qt5Widgets.dll was not found. Reinstalling the program may fix this problem.
The downloaded software was packaged as a .rar file, which I converted online previously. Installed WinRar to see more details, and it gives 3 errors while unpacking: twice an `Unexpected end of archive` and once `Checksum error in DGUS_TOOL_V8.0.1.1_Eng\DGUS_TOOL_V8.0.1.1\QtWidgets.dll. The file is corrupt`.
Tried again with "keep broken files" enabled, and got the error `C:\file\location\Qt5Widgets.dll is either not designed to run on Windows or it contains an error. Try installing the program again using the original installation media or contact your system administrator or the software vendor for support. Error status 0xc00007b.`
Pretty clear that specific file is broken or corrupt.
Tried to run it with DWIN software V7.3 as well, but cannot find how to export as .ICL file, which is necessary according to me.

Sebazzz, 2020-11-16:
No, the DWIN 8 software must be used.

Tillema, 2020-11-16:
Great! This one worked. Will start fiddling around with the software first. Thank you!

closed 2020-11-16 by Tillema