CR6Community/CR-6-touchscreen #19: Touchscreen freezes when selecting Print option

Dallagen2, 2020-12-04:
Flashed the marlin rewrite on printer and the community touchscreen firmware, auto level works fine, but pressing print freezes the touchscreen until I power cycle

Sebazzz, 2020-12-04:
Is this after selecting a file? Please provide logs.
Is this *before* selecting a file? Please flash the touch screen firmware.

Dallagen2, 2020-12-04:
This is before selecting a file, and the touch screen firmware has been
flashed, just redid it to make sure.

Sebazzz, 2020-12-04:
Sorry, then you didn't flash it properly (or didn't take the correct file).

Please check the updated instructions here: https://github.com/CR6Community/CR-6-touchscreen/blob/extui/src/README.md

Dallagen2, 2020-12-04:
found the problem, combination of me being a dumbass and github having poor UI

GitHub hid release 4 by default since it's tagged pre-release, so I flashed the pre-release touchscreen with release 3 firmware, updated printer FW and now it's fine

closed 2020-12-04 by Dallagen2