CR6Community/CR-6-touchscreen #24: EEPROM Default settings

chriskoups, 2021-01-02:
I am not sure if the default settings are saved in the screen (when you call M502) but the E jerk is not set. In default must be 5.0.

Sebazzz, 2021-01-02:
Can you elaborate on that?

chriskoups, 2021-01-02:
Morning, yes i can.
I flashed the complete RC5 with the screen and firmware on my CR-6 with MB 4.5.2. Then i press the "restore settings" on the screen so EPROM will be reset.
While i was printing, i notice some little bumps every ~10mm which were weird. I try to calibrate linear advance and when it was not 0 it was making a jittery noise.
I look the EPROM and the command M205 was outputting E0.00 which means the extruder is trying to achieve the speed with maximum acceleration and if they are high (i think are set to 5000?) Is cogging.
I fixed the problem by passing M205 E5.0, which is the same as the original EPROM.

chriskoups, 2021-01-02:
Sorry, My bad! I am using EEPROM editor on octoprint and mess up the M205 command making the E0 despite set to 5! My bad....

closed 2021-01-02 by chriskoups

Sebazzz, 2021-01-02:
This is I think exactly the issue @Ranney1 ran into 😄

Ranney1, 2021-01-02:
OMG, my stupid action is following me 😊🙈👌

Sebazzz, 2021-01-02:
> 
> 
> OMG, my stupid action is following me 😊🙈👌

Wij zijn betrokken, maar niet onfeilbaar 😀