CR6Community/CR-6-touchscreen #25: [FR]  Please define and implement a display convention that makes it obvious when it is possible to tap a displayed value to change it

Thinkersbluff, 2021-01-06:
I really love the feature-richness of this UI. 

 Today, while using the Filament Extrude menu, I "discovered" that I could tap on the displayed nozzle temperature in the footer and get a pop-up that let me change the target temp. 
Until then, I had been navigating back & forth between the Filament Extruder menu and the Control/Temperature menu to make that change.
While I am happy to discover the "Easter Egg", I do not trust myself to remember it each time ("in the heat of the moment", as it were).  I am also left wondering what else I can tap...

Please consider - within the many constraints imposed by DWIN - how you can make it obvious to a user that tapping on a value will popup an edit box.
e.g. I used to think that if a box appeared around the value, it could be tapped to modify, but evidently that is not true, because the MOVE menu displays axis positions in boxes and those cannot be tapped to modify.

I recognize that whatever convention you adopt will likely require a purge of all screens to ensure/implement consistency, and the convention will need to be documented, to ensure consistency is preserved, moving forward, so perhaps this FR can be considered within the scope of the proposed UI Menu Rework.  If something could be done in the short term, though, it would be appreciated. (e.g. maybe text or a redundant control on the FIlament Extruder screen to set Nozzle Temp?

Cheers!

Sebazzz, 2021-01-06:
I agree we need to be consistent, however, I think that this little easter egg can be left as the easter egg it is as long as we don't repeat this trick 😀 I think the UI gets very "restless" if we start adding visual cues of boxes everywhere.

Leaving this to @grobux for his opinion. 

> e.g. I used to think that if a box appeared around the value, it could be tapped to modify, but evidently that is not true, because the MOVE menu displays axis positions in boxes and those cannot be tapped to modify.

Those can, actually. This works fine at my side. A menu appear, and the printer responds.

Thinkersbluff, 2021-01-26:
> Those can, actually. This works fine at my side. A menu appear, and the printer responds.

I don’t know if that was a change, but yes I can also now tap to modify the x-axis positions on the Move menu.  Looks like Z Offset on the Level menu is now the only boxed parameter I can not tap to modify.  Making that also tap to modify would make perfect sense, when we flash and restore the factory setting of 0.20 and we know our offset was actually 0.08 before the flash, for instance.

Seems a boxed value with a light grey background is almost a defacto standard format for a “Tap to modify” visual cue.

If we decided to make that a consistent standard, it would be easy to explain in the release notes.

Thinkersbluff, 2023-03-13:
So, as the guy who first opened this, I am now going to allow myself to close it.

I did try to implement a fix for this, by positioning a white box behind any field that could be tapped to change.
Unfortunately, some of the fields I wanted to make variable had to be placed where the portrait-oriented display did not provide enough space for that box.  Rather than shrink the text to fit those boxes, I decided to give up on this idea.

closed 2023-03-13 by Thinkersbluff