CR6Community/CR-6-touchscreen #27: Allow 180 degree rotation of the display

rancid-racer, 2021-01-12:
I would like to be able to rotate the image 180 on the initial orientation for my CR-6 so that the SD Card Slot is at the top of the board instead of the bottom.  This allows for an opportunity to reprint a back cover and make it easier for updates without removal of the panel.

ritchiedc, 2021-01-12:
I used a 15cm micro sd extender. It was a tight fit. I had to cut a notch in the back cover and pulled the display off of the bracket to insert the cable but it works well. I didn't have problems sticking the display back on but that could be an issue.

Sebazzz, 2021-01-12:
It is actually a pretty good idea, also for development. Saves me from buying an extender that may or may not work (some really do not work).

Note that the initial boot screen until Marlin actually is online and loaded the settings from EEPROM will still be in the original rotation (so essentially the boot screen will be upside down).

Moving this to the Marlin repo.