CR6Community/CR-6-touchscreen #30: DGUS update to v40 makes touchscreen unresponsive

wattie2k1, 2021-01-29:
Since the latest commit to DGUS update the touchscreen will not respond to inputs anymore. Flashing finishes successfull, the printer boots up fine and connects to octoprint. First touch on the touchscreen gives a beep, more touches don't give any feedback anymore. 

Flashing the touchscreen back to previous version with DGUS v35 makes the touchscreen responsive again and works as expected. So I suspect a problem with the DGUS update to v40.

grobux, 2021-01-29:
Hi @wattie2k1 and thanks for your report.
I may have updated DGUS a bit quickly, without testing enough.
I check further on my side.

grobux, 2021-01-29:
Issue confirmed, investigating.

grobux, 2021-01-29:
I have more test to do before upgrading the OS. Even v36 crashes.
It seems we have to patch the ICL files with a third party tool..,
DGUS software doesn't do it and doesn't seems compatible with patched ICL files.
So I take the simple route for now, and revert back to v35.

I promise that next time I will do more testing ! 😁

closed 2021-01-29 by grobux