CR6Community/CR-6-touchscreen #31: [bug] Fonts Clipping

coreyostrove, 2021-02-09:
I noticed some clipping of certain letters on the stock creality touchscreen LCD. See the attached photo for an example with the letter g.
[image]

grobux, 2021-02-09:
Hi @coreyostrove  and thanks for taking the time to report this issue.

We have this issue since the redesign of the firmware, but it hasn't been addressed since. Your message may be an opportunity to get back to it.

One solution could be to make lowercase letters small capitals. That way, all letters will stay in their available space. It will also harmonize with text in the pictures.

Some people are experimenting this idea on Discord, but DGUS isn't helping making nice fonts.

Sebazzz, 2021-02-10:
@Thinkersbluff is also enjoying the wonders of DWIN font rendering 😅

Thinkersbluff, 2021-02-10:
The DGUS font clipping behaviour takes many forms with a variety of contributing factors.  The issue in this photo seems directed at the clipping of descenders.  I have been exploring ways to improve on that. Aesthetics are a large part of resolving this issue. The current implementation may yet turn out to be the best balance, but I have not given up trying.
This is a different font but these two photos demo one option.  Using FontForge, the 5 glyphs with descenders can be edited to bring the descenders up to the baseline, for instance.  Another option is to use small caps.
[image]
[image]

Notice that if I try to take advantage of increasing the font size to take advantage of the descenders being raised, then numbers don’t fit fields as well.  The modified font may also not be as pleasing to the eye.  I am hoping small caps is a better answer, but it still has its issues.  Just scaling the caps down gives serious anti-aliasing problems with the current font. Other fonts are getting clipped on left or right side of some characters on some screens.  The tools are not well documented, so it will take time and patience to finesse this issue. I will share any reasonable looking fonts I can.

Thinkersbluff, 2021-02-10:
[image]

I find this font to be a reasonable alternative to the B612Mono font being used now.  The sides of some letters still get clipped in some display fields, for reasons I have not yet understood, so it is not “perfect” but I can do no better for now.

Sebazzz, 2021-02-10:
I suspect it has something to do with the height of the fields, but I can't be sure. There is this weird vertical height and horizontal weight parameter in the DWIN editor.

Thinkersbluff, 2021-02-10:
[image]
[image]

Thinkersbluff, 2023-03-13:
The Re-Factored Version1.1 released 13 March 2023 implements a new HZK file which solves the problem with descenders, though it also shrinks the message font slightly in the process.

closed 2023-03-13 by Thinkersbluff