CR6Community/CR-6-touchscreen #33: “S” hidden when printing time has double-digit minutes and seconds

Thinkersbluff, 2021-02-22:
Running Rel 6 Pre 4 self-compiled on 21 Feb.

Once the minutes and seconds both show double-digits, the S disappears.  If possible, please increase the field size or shrink the font or reduce the space between minutes and seconds?

[image]
[image]

Sebazzz, 2021-02-22:
Thanks, bbc868e should fix it.

closed 2021-02-22 by Sebazzz

Thinkersbluff, 2021-02-24:
Sorry.  24 Feb extui Rel.6 Pre 4 not fixed.
[image]

Sebazzz, 2021-02-25:
For [issue #114](https://github.com/CR6Community/Marlin/issues/114) in the main repository @ztakis is working on a design. I've suggested to him to move the stopwatch icon a bit to the left to accommodate for an even wider label.

Thinkersbluff, 2021-02-27:
> For [issue #114](https://github.com/CR6Community/Marlin/issues/114) in the main repository @ztakis is working on a design. I've suggested to him to move the stopwatch icon a bit to the left to accommodate for an even wider label.

Well, the good news is you don’t need a wider field and you don’t need to move the stopwatch icon.  
[image]
 I added a second zero to the Textinit parameter on screens 36, 37 and 39.

Now, I am into a 7-hr print, but will try removing the “s” from that field on the next iteration.  
[image]

Sebazzz, 2021-03-01:
Got it, string length was set to 6 at the Marlin side.

commit bbc868ef (2021-02-22): Slightly increase field width of print timer to prevent clipping - fixes #33

Set HorizontalSize from 18 to 17, and increased the field width a bit.
Done this on printing, paused and finish screens.