CR6Community/CR-6-touchscreen #35: "M117 Message"

areaknight99, 2021-02-26:
Hello, I'm new to this community firmware and trying to install it on my CR-6 SE. I think I got everything correct but after flashing the touchscreen, the "M117 Message" shows up. When I touch the "Print" the screen froze and I cannot touch anything anymore, it just froze.
[image]

Thinkersbluff, 2021-02-27:
This is typical behaviour when the Motherboard and the Display are not both flashed successfully.
The release notes are quite verbose in their guidance because a lot of users report problems getting the display or motherboard flashed.  This is not a firmware issue.  When you have both matching firmwares correctly flashed, you will see ".... Ready", where "..." is the name of the firmware release you have loaded.

areaknight99, 2021-02-27:
> This is typical behaviour when the Motherboard and the Display are not both flashed successfully.
> The release notes are quite verbose in their guidance because a lot of users report problems getting [...]

Where is the file so that I can flush the motherboard?

Thinkersbluff, 2021-02-27:
> Where is the file so that I can flush the motherboard?

There are several versions of the Community Firmware presently being used by the Community members. 
The last full release was at [version 3](https://github.com/CR6Community/Marlin/releases/tag/v2.0.7.1-cr6-community-release-3).

There was [an alpha version 4](https://github.com/CR6Community/Marlin/releases/tag/v2.0.7.1-cr6-community-release-4-alpha), which was the first version using a display firmware completely rewritten for this project.  That motherboard firmware was based on an unreleased version of Marlin, so it was labelled an "alpha" release. 

A stable "beta" firmware was released at [version 5](https://github.com/CR6Community/Marlin/releases/tag/v2.0.7.3-cr6-community-release-5-beta), re-based on the Marlin bugfix-2.0.x sources which will eventually become Marlin v2.0.7.3.  Since the formal release of Marlin 2.0.7.3 continues to be deferred by the Marlin team, beta release 5 was never formally released, but has since been evolved into a series of pre-release versions 6.

The most recent versions are [these pre-release development versions 6.](https://github.com/CR6Community/Marlin/releases)

I recommend you review the release notes for these versions and decide with which mix of features and risk you are most comfortable.  You will need to know which motherboard version is in your printer, to select the right zip file.  The display firmware is a zipped file inside the motherboard zip file.

areaknight99, 2021-02-27:
> There are several versions of the Community Firmware presently being used by the Community members.
> The last full release was at [version [...]

 Thank you! I figure it out that I didn't flush my motherboard.

closed 2021-02-27 by areaknight99