CR6Community/CR-6-touchscreen #53: ICL tool in english

nicwilson58, 2021-04-05:
Apologies if this is in the wrong place , it is not obvious where to put such a question.

I have the V8.0 tools from the supplied link and its fine.  However I am unable to find the English version of the ICL tool as shown in the example.   I have found the "AllTool.exe" which is the Chinese version and it does work OK so really an issue,  it might be able to set it to English but its not clear either.  Is there an Englihs copy or a way to set the Chinese versions language?

nicwilson58, 2021-04-05:
found it

closed 2021-04-05 by nicwilson58

Sebazzz, 2021-04-05:
From the DWIN tool itself, as you've found out 👍 
I'm not sure how and if it is possible to run the tool standalone with the correct language.