CR6Community/CR-6-touchscreen #61: [FR] Color legend on ABL screen

lukemaurer, 2021-05-15:
Every so often, the CR-6 SE community Facebook group gets a post from someone seeing all green numbers on their ABL screen, not realizing that green means very good, and it turns out they've been struggling trying to get the numbers down even further. I think it would help if the ABL matrix added some context by showing the color scale somewhere. (A visual indication that anything less than red is perfectly okay would be even better, but I'm not sure how to do that on an already-packed screen. Maybe if the colors just ran from green to yellow to red?)

[image]

Sebazzz, 2021-05-15:
The numbers don't say anything. Red doesn't necessarily mean bad, it just means relative much from zero. It is visualisation, nothing else.

Met vriendelijke groet,
Sebastiaan Dammann

[CB746FCB-2ED8-4FAE-A30A-C140A9BD1BAA]<https://emea01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F262648%2F118371908-aa391b80-b5a6-11eb-9e95-a1403f924456.jpeg&data=04%7C01%7C%7Cc7afdea9b3264707b18a08d917c4d69a%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637566956439801556%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=pb7mVr9%2BRMyptFjcFi6f0TekaZUAFEKMWNPmVN7%2BD9E%3D&reserved=0>

KAPTOHKA, 2022-09-11:
I would like to make the color of the numbers not depend on the total zero, but would depend on the curvature of the table...
For example:
We have a table grid, where the maximum figure (+0.234mm) and the minimum (-1.234mm)...
The middle between these levels will be (1.234+0.234)/2=(-0.734mm)...
To the resulting figure, add / subtract half of the maximum allowable curvature of the printer table, for example 0.3mm / 2 = 0.15mm ...
The number (-0.734mm) is painted white.
The numbers ((-0.734mm) + 0.15) = (-0.884mm) and above are painted in the red spectrum.
The numbers ((-0.734mm) -0.15) = (-0.584mm) and below are colored in the blue spectrum.
Thus, the curvature of the table will be visible, and not a deviation from zero ...
I suspect that after measuring the curvature and calculations, it will be necessary to somehow redraw the screen with the numbers of the curvature of the table so that new colors are applied ...

@Sebazzz  can you please tell me how to implement this?
and in what files is the processing of data on the grid?

Sebazzz, 2022-09-12:
@KAPTOHKA I believe this is what you are looking for:

https://github.com/CR6Community/Marlin/blob/25fb55803deee3233ee440a7330f982a64fb720d/Marlin/src/lcd/extui/dgus_creality/DGUSScreenHandler.cpp#L920-L1009

Thinkersbluff, 2023-04-20:
The colour scheme for the bed mesh is actually implemented in the motherboard firmware, not in the display firmware.
Any Feature Requests suggesting changes to that design should be raised on the motherboard firmware repo.

FWIW:
The current design is based on a heatmap visualization, which is common in 3D bed mesh visualizers. Hobbyists already familiar with other visualizers will recognize the scheme without need for an on-screen colour-key:
- Red = hot = close to the nozzle. 
- Yellow = warm = between Red and Green
- Green = close to neutral
- White = neutral
- Green (negative value) = close to neutral
- Cyan (negative value) = cool = between blue and green
- Blue (negative value) = cold = far from the nozzle.

Since it is the homing probe point at the center of the bed which establishes the overall height of the bed mesh, just before printing, all of the values in the bed mesh are measured and stored relative to the measured height of that same bed center point, as measured just before abl probing begins.  
The colour scheme is also therefore also defined relative to that bed-center probe point.

It is an unfortunate fact that primary colours like Red, Green and Yellow also tend to have socially significant meanings like "Stop/Caution/Go" or "Danger/Caution/OK", which may cause some folks to think the mesh colours somehow connote "good" or "bad" results.   

For Users whose beds are particularly warped, the bed mesh colour coding will produce a "rainbow" effect.  The ABL function itself will contentedly follow the mesh, but since it must interpolate the bed height between points, some intermediate heights may be a little "High" or "Low", since the physical curvature of the bed may not follow the mathematical model used by Marlin to approximate that curvature.  

"Fiddling" with the bed screws, to modify the warp on the bed is one option, which in turn is likely to manifest as a more uniform "greening" of the mesh values.  The more consistently "trammed" the bed to the nozzle path, the more consistently will the bed mesh readings be green (or white if =0.000.)