CR6Community/CR-6-touchscreen #63: CR-10 Smart stock display orientation

badbod, 2021-07-24:
HI, Small problem,
I flashed the latest firmware from community 6.1 release and while it is working, the display is landscape and squidged up to one side of the screen, and it is upside down..

I assumed it would be 90 degrees rotated (portrait) but it seems to be 180 degrees rotated (landscape).

Is there an easy way to rotate it?   even though in landscape it is small and squidged into one side of the screen it is actually usable  and working. just upside down.   Cables length and mounting slots etc it won't mount upside down :-(

something in source somewhere I can modify to rotate it?

I noted at first it would not work at all, I had dgus 4.1,  adding the kernel files to change it to 3.5 fixed that.

going to try running both 12864 and upside down stock at same time now, wish me luck.

Thanks.

grobux, 2021-07-24:
Hi,

I don't know much about the CR-10, but you can try to change the screen rotation setting on the screen profile.

- Open DGUS and select "profile generation",
- Open the screen configuration (https://github.com/CR6Community/CR-6-touchscreen/blob/extui/src/DWIN/DWIN_SET/T5LCFG_272480.CFG),
- Change the display direction to the desired parameter,
- Save the CFG and flash your screen with it.

[image]

badbod, 2021-07-24:
Ill certainly try, thank you. I have not done anything with meddling with the dgus, Marlin im fine with  :-)   ,  but will give it a shot,   

edit that file with dgus, save the config,  just copy that config over to a ready DWIN_SET ?

grobux, 2021-07-24:
> edit that file with dgus, save the config, just copy that config over to a ready DWIN_SET ? Or just place that one file on its own in the DWIN_SET folder it will flash it?

Both should work for CFG updates. You can flash files individually, as long as you keep them in a DWIN_SET folder at the root of your SD card. On the blue screen in flash mode, you should only see the file types you're flashing (1 CFG, 0 for others, if you only push an updated profile).

Sebazzz, 2021-07-24:
Yup open the official firmware cfg, then change the orientation parameter.

Met vriendelijke groet,
Sebastiaan Dammann

badbod, 2021-07-24:
Thanks  I downloaded dgus 7.3  for dgus II and am now trying to find my way to the screen as in your piccy.  LOL,  total newb in da house, sorry.  Ill find it though.  Nice to know I can just flash the CFG , takes a load off dealing with trying to compile all of it  LOL

Sebazzz, 2021-07-24:
We use DGUS 8.x, there should be a link in the documentation.

Met vriendelijke groet,
Sebastiaan Dammann

badbod, 2021-07-24:
ok, yes, link in docs.  I edited it, and 1 step closer, but...    the upside down is how it was original, and the one on it's side is new. the upside down one still works,   think maybe I didnt tick something? to apply to all   err  'screens'  hmm,  anyway, A minor triumph at my first DWIN dgus hacking attempt   :-)

[image]

badbod, 2021-07-25:
hmm,  started flashing and went blank, now I cant seem to get any life out of it.
  I tried Troubleshooting-experimental-DWIN_SET_format_kit.zip ,  just stays blank. How long shoudl I leave it to make sure it finished flashing? I left it about 10 minutes. Then tried to flash both original and community, I dont get the blue progress screen, wonder if I broke it? 
 was not unpluggin it  or resting it on any metal, just swapping the sdcard in and out while trying options from the preferences config.

is there a full reset procedure I can follow?

grobux, 2021-07-25:
Strange. this screen is quite hard to brick. The bootloader isn't directly accessible through the firmware flashing procedure and would require to solder a serial interface directly on special pins of the motherboard.

The format kit you try is, I guess, the one that set all the memory bits to 0. So it's some kind of memory reset.

You mentioned that you took care to not let the motherboard on a metallic surface, so I don't go further on that way.

You can check the cable between the motherboard and the screen, maybe it isn't well plugged on either side or is damaged.

In most buggy scenarios I've encountered, I've flashed an old official firmware (like v1.0.2) to step back before restoring the community firmware. But I never got a totally blank screen, at least a freeze on the boot screen or something like that.

Maybe your SD card has issues. You may try to reformat it and/or try another card. The screen is EXTREMELY picky concerning memory cards. Flash all the firmware files to be sure to reset all the memory sectors.

Good luck, keep us in touch.

badbod, 2021-07-28:
I had some issues with it blasnk when I first hooked it up thto the DKR board, then after some fiddling it started working,  I will connenct the original board to a 24V supply and try that on the workbench.,  then I need to look at the mounting for the SKR board as I cant get the screw in next to the display socket, so maybe the metal pressfit thread is touching something it shouldn't.  

I will do that this weekend and get back to you with my findings. I think with the screen orientation and sizing, I wont be using it for now. As you guys have announced support I will wait and try the display when it fits the screen and is not upside down  :-)   

Using a CR-10 Stock (12864) I have managed a couple of prints,  changed leveling to UBL and generally can do everything I have not been able to do so far with the stock board and firmware. ripped out the silly wifi unit and changed over the USB etc,   raspberry pi fits nice and works too. Actually usable now.   3 weeks from arrival to getting to put filament in cos it would not even home properly,  and Creality support seem to think I should use an app and why am I pressing things on the LCD screen anyway of course it will home Z at -5,-2 if you do that. What? nozzle crashed into bed, sounds like you tried to raise Z from the LCD,  LOL. 

    Probably should not show them my - pimped out Ender V2 BLtouch 10X10 UBL with full bed coverage +50 on X, dual Z dual driver.  Dual 20mm Nema steppers, dual BMG (clear)  direct drive (does not work as well as it sounds, but looks great :-), CR-10 V2 Z sliders (red)  all the auto everything Marlin bugfix weekly builds - monster.

anyway, one question.
What files do I need to compile Marling bugfix using the SKR-CR6 board with a 12864 display only?   I am not sure which files I will need so I can set 
  #define MOTHERBOARD BOARD_BTT_SKR_CR6 
  #define CR10_STOCKDISPLAY

 and it will compile. I already guess src/pins/pins_BTT_SKR_CR6.h is one,  I think there is a bit more to it though?

badbod, 2021-08-05:
ok, much playing later, the unit is dead.
 5V in 60ma power draw.  Checked the SKR board where the screw does not align, the metal threaded insert is a tad proud of the plastic but is not close enough to anything to short.  No amount of push/pull etc causes a short and I can see nothing for it to short to. That it blinked off during flashing I think is co-incidence, so far the most power on time the display it has had is flashing, so I think it just failed during and not as a result of.
   
 Found dodgy sd card slot connections and re-soldered them, had it under the microscope for a while with meters checking components, all fine.  High power light though screen edges and rear, cannot see any LCD activity, not just missing backlight then. 

it is 20ma below spec and no amount of play changes that so I think it is just dead and flashing had nothing to do with it.

Until a working UI is rolled out for this display it is pointless to replace. Ill stick with the 12864 for now. I would like to take the latest Marlin bugfix and build for SKR CR6/CR-10 Stock, ill post in discord see if any one can help with that.  However, your current ver 6.1 is working flawlessly so not in any particular hurry I guess :-)

Printer is doing fine @ 70mm/s with 800 acc, 700 feed, JD .035, retracts 6000 acceleration,  6.5mm 25r/45ur mm/s, UBL 7x7, 15 inset no margin, .22 z-offset.  no LIN , hybrid x/y600,z150,e30. Fast and clean prints   HURRAH.  I have it in an enclosure so YMMV with these settings.

Thanks @Sebazzz, @grobux  for your help.  Closing this one out.

closed 2021-08-05 by badbod