CR6Community/CR-6-touchscreen #64: Updating via uart

zgdump, 2021-08-19:
Hello! 

I don't have an sd card for updating the firmware and I decided to implement the update via uart. This possibility is present, according to the documentation. It seems like it's enough to send 32kb to RAM, and then send a write command to flash memory.

[Documentation](https://github.com/CR6Community/CR-6-touchscreen/blob/d4afcdadeefcfb40e28ba7497001602b9a912ba5/doc/vendor/T5L_DGUSII%20Application%20Development%20Guide20200902.pdf). Took the information from pages 16, 36, 48

But it doesn't work. The screen reports OK, but the memory is filled incorrectly - after a reboot, it shows a black screen and does not respond. I've been sitting here all day, rereading the documentation, experimenting. But to no avail.

If anyone helps , I will be grateful!

Code. [DwinUpdater.zip](https://github.com/CR6Community/CR-6-touchscreen/files/7012920/DwinUpdater.zip)

Sebazzz, 2021-11-29:
I think @grobux did some testing on this.

zgdump, 2021-11-29:
> I think @grobux did some testing on this.

If someone helps me, I will be very glad. During all the experiments, I received a stable update, but 32.icl is only partially updated. Only the first background is present, there are no others. For how many weeks I've been struggling and everything is useless

zgdump, 2021-12-02:
> If someone helps me, I will be very glad. During all the experiments, I received a stable update, but 32.icl is only partially updated. Only the first background is present, there are no others. For [...]

I figured it out, the question is closed

closed 2021-12-02 by zgdump

Sebazzz, 2021-12-02:
Can you enlighten us? 😁

[X]

Met vriendelijke groet,
Sebastiaan Dammann

Closed #64<https://emea01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2FCR6Community%2FCR-6-touchscreen%2Fissues%2F64&data=04%7C01%7C%7C73a6c4eba0a64f73af8208d9b5b8e107%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637740627921868417%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000&sdata=6X3dt9n1Vb%2FxMU1xBODBTjG7vktGUc9x9TH%2FwdGQZ3w%3D&reserved=0>.