CR6Community/CR-6-touchscreen #65: please flash tft screen

joshgebbie, 2021-10-12:
I turned off my machine after the last print and when I turned it on again it had an error "Please flash your TFT screen: version mismatch - build 61166 found but expected 61" I reset factory settings and re flashed the display and got the same error.

joshgebbie, 2021-10-12:
I tried using the last firmware and the error message changed to "Please flash your TFT screen: version mismatch - build 60 found but expected 61" so there was a change then when I reinstalled the newest firmware I got the same message "Please flash your TFT screen: version mismatch - build 61166 found but expected 61". I cant find a post explaining how to clean format to clear flash memory I barely know what I'm doing now.

StephenBrough, 2022-06-06:
Have the same issue, tried flashing the screen about 10 times. Also tried flashing older community firmware and then 6.1, also tried flashing official firmware and then 6.1. Will try compiling out that check next.