CR6Community/CR-6-touchscreen #66: CR-6 SE Screen Gray on boot

osxfr33k, 2021-11-08:
I see one little green dot must be dead pixel anyhow now I see 3 files inside the kernel update folder am I suppose to put all three files into the dwin-Set folder?  I wish the instructions were clear about this.  I see one UI file and two OS files in the kernel folder.  My Motherboard version is Era v1.1.03.

ritchiedc, 2021-11-08:
Per the instructions in the readme included in the distributed zip, "If you have never run the Community Firmware release 5 beta or higher and you have a Creality CR-6, move the files in the "CR-6 kernel upgrade" directory to the DWIN_SET directory."

Sebazzz, 2021-11-09:
Yes, that's right. Might want to add an additional file to document this in the folder.

osxfr33k, 2021-11-09:
OK so I never updated I see my Screen version is 4 so its downgraded v3.5?  Anyhow there are two OS bin files in there a v10 No RX8130 and one that is v11 RTC RX8130 and a 3rd bin "T5L_UI_DGUS2_V35_20200730".  So put all three in the "DWIN_SET" folder along with the rest of the files already in that folder?   Also is this downgrading from the current version 4.0 to 3.5?  Thanks!

closed 2021-11-10 by osxfr33k