CR6Community/CR-6-touchscreen #70: touchscreen reboots when touched

tcables, 2022-02-17:
Hi, 

my touch screen is rebooting all of the sudden when touched. I'm running latest 6.1 firmware for 6+ months just fine. nothing changed / updated. Touch, reboot, touch, reboot, touch touch touch wont boot. power cycle, same issue. I also tried flashing the ' Troubleshooting-experimental-DWIN_SET_format_kit.zip' and I only get 000's. Any help troubleshooting or a reset would be much appreciated. Thanks!

Sebazzz, 2022-02-17:
Try reseating the cable, completely. From mainboard to the screen itself. I do not believe this is a software issue, but probably something like an intermittent ground failure. On the CR-6 the cable tends to get pinched behind the drawer and cause trouble.

tcables, 2022-02-19:
Thank you for the suggestions. I took apart the bottom to reveal the motherboard and cables. I wiggled and reseated the touchscreen cable on both ends. This did not seem to resolve the issue. i also traced the cable and saw no tears, pinches, or degradation. Is there a different way to flash the stock firmware, or reflash the community firmware? Thanks!

Sebazzz, 2022-02-22:
Sorry, I have no further suggestions. This sounds like hardware failure. Are you sure the touch screen is rebooting and not Marlin itself? Have you confirmed this with a serial monitor?

> I also tried flashing the ' Troubleshooting-experimental-DWIN_SET_format_kit.zip' and I only get 000's.

You should not get that, so I'd follow the normal flashing procedure with these files ready.

tcables, 2022-02-23:
I'm a bit confused, what is the normal flashing procedure? The guide says to just put the files on a 16gb fat32 sd card and insert into the touch screen, then boot - which is what gave me those 000's

ritchiedc, 2022-02-23:
The readme states you have to use a micro sd card 16gb or smaller formatted as fat32 with an allocation unit size of 4096. It also says you need to copy the three kernel upgrade files to the dwin_set folder if you haven't flashed the community firmware before.

It is better to ask these kinds of questions in the discord for the cr6 community firmware.

tcables, 2022-02-23:
Oh, my apologies.

We can close this and then I can reach out on discord to continue.

closed 2022-05-10 by Thinkersbluff