CR6Community/CR-6-touchscreen #76: Stuck in boot loading

jayse2006, 2022-04-19:
Printer says the bootloader is confused. attempted to reflash the firmware but it wont do anything other than display the start screen.

Thinkersbluff, 2022-04-19:
This is not enough information for me to help you.
That error message says try another card.  If that is not helping, you may do better raising this as a discussion in the Discord channel.

jayse2006, 2022-04-19:
I’ve tried a different card and even tried booting without the SD card in it. It’s been left on over night to give it time to try and boot. It has worked for months with the firmware and now this pops up

jayse2006, 2022-04-19:
[image]

Thinkersbluff, 2022-04-19:
> It has worked for months with the firmware and now this pops up

If the printer worked on this firmware for months, then it does not sound like this is a bug in the firmware.  It sounds more like the firmware is working correctly, but the motherboard bootloader is not.  

I recommend that you come discuss your hardware issue on the Community.

 https://discord.gg/yDY6pb2BQS

closed 2022-05-10 by Thinkersbluff