CR6Community/CR-6-touchscreen #78: Strange screen is displayed on Touch Screen firmware flashing.

junonara, 2022-05-10:
I just ordered an additional touch screen for the CR-6 SE and it was delivered.

While flashing the new CR-6 SE's touch screen using the 61F_RC_290422_v1 released by CR6Community/CR-6-touchscreen, near completion, a white noise-like screen appears.
[image]

It looks like the flashing is almost done right before the white noise display.
[image] [image]

The new touch screen uses the same PCB as the old LCD. However, flashing completion is indicated differently. However, if I wait a few tens of seconds and turn on the CR-6 SE, it operates normally (displaying icons, selecting menus, etc.).

First, to check the kernel version, I checked the kernel version by flashing using an empty DWIN_SET, and at this time, the normal completion screen was displayed. After flashing using the contents of DWIN_SET, a strange completion screen is displayed.
[image]

On the touch screen I had it shows the normal completion screen. It is strange because it is a symptom that appears on a new touch screen that uses the same PCB.

scratchfury, 2022-05-10:
The white noise screen occurs when the screen has to change orientation.  If you power it off and back on, it should display normally in the new orientation.  If it was a new screen, it was probably in orientation 0 and the CR-6 uses 270.

Sebazzz, 2022-05-10:
Interesting. I've never seen noise!

Met vriendelijke groet,
Sebastiaan Dammann

Thinkersbluff, 2022-05-10:
I know the 4.5 calibration process is performed in landscape orientation. I wonder if they left it in that orientation at the factory. 
Glad it is ok now.

junonara, 2022-05-11:
If the calibration or rotation setting is wrong, the same phenomenon should appear on two touch screens of the same PCB version, but this phenomenon appears on the touch screen purchased later. Of course, I think that even the same PCB may contain HW with different factory settings or slightly different physical signal timing.

Once the white noise screen is displayed, it is not possible to know exactly when the flashing is completed, but if I turn it off and on after about 30 seconds, new touch screen works normally, so it is in use for now.

Thinkersbluff, 2022-05-11:
Actually it is not the hardware, it is the operating system, that makes two of these screens behave differently.  At DGUS2 v4.5, the calibration changed.

junonara, 2022-05-11:
The kernel version shown at the first flashing using empty DWIN_SET was T5L1 DGUS2 V4.5 2022.01.13, and when flashing by copying the contents of the DWIN_SET folder, a white noise screen appeared and flashing again with T5L_UI_DGUS2_V45_20220105.BIN.

Thinkersbluff, 2022-05-11:
I missed that detail, in the photo with the white noise.
Why did you add a DGUS BIN file to the DWIN_SET folder when flashing the refactored DWIN_SET?

junonara, 2022-05-11:
After flashing DWIN_SET (without kernel), a noise screen appeared and I thought it was a kernel problem, so I flashed with the kernel again.

Thinkersbluff, 2022-05-11:
Sorry. I have no other knowledge to share, here.  I am baffled.

if you remove the CFG file from the DWIN_SET folder and flash again, the noise screen may not appear.  It is the CFG file that commands the screen hardware to change orientation or mode.

Sebazzz, 2022-05-11:
Yes and the orientation is actually overridden via a command after the firmware initializes.

Met vriendelijke groet,
Sebastiaan Dammann

Sebazzz, 2022-05-11:
> After deleting T5LCFG_272480.CFG from the DWIN_SET folder and flashing, the completion screen is displayed normally. Thanks!!
> [IMG_20220511_161217_DRO]<https://nam12.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F96027590%2F167791725-978bc4e2-570e-40c5-b0f8-49a297d753e8.jpg&data=05%7C01%7C%7C3b2453394ad64967313808da331eb1d2%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637878504162374192%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=lbSqq4GM%2FHZzmFXDH5c4FKG6jsG5kwzRQumeZ0N8Vkw%3D&reserved=0>

It might have something to do with other parameters in the CFG file. For instance: capacitive vs resistive touch screen is also configured there I think.

Met vriendelijke groet,
Sebastiaan Dammann

Thinkersbluff, 2022-05-11:
> After deleting T5LCFG_272480.CFG from the DWIN_SET folder and flashing, the completion screen is displayed normally. Thanks!!

Glad to hear it.
I will leave this issue open for a while, so that others can find it more easily.  

@Sebazzz - There is so much online info now about flashing CF6 and of course  no mention of this DGUS2 calibration issue in the June 2021 release files.  I expect we will be hearing more and more from folks receiving new printers and replacement displays who have not yet heard about the refactoring of DWIN_SET or of the calibration issues with DGUS2 at 4.5+

Are you thinking of using the Announcement or Releases channel, to “promote” this change from RC to full release?

Will you need other user feedback on this RC before you decide next steps?

Sebazzz, 2022-05-11:
It is not a matter of deciding, because I think it is ready to go. With a note that this primarily meant to resolve issues for users with newer hardware, and not an upgrade for existing printers. The only question remains whether we have documented enough.

Thinkersbluff, 2022-05-12:
Cursed with an Engineer's brain. I still have questions. 
Will you pull this release into the main board zip files (e.g. as a maintenance release, like 6.1.1), or leave folks to discover the refactored Touchscreen firmware here?

Thinkersbluff, 2022-05-30:
UPDATE: It may be coincidental, but this issue has not been reported again, since we stopped using the build script to assemble DWIN_SET for the release. That script renamed ICL files 13 and 14.  Maybe that name change "bothered" this new version somehow.

Thinkersbluff, 2022-06-11:
Since there have been no more reports or observations of this problem since we stopped renaming files 13 and 14, that is now the recommended standard practice. This issue is closed.

closed 2022-06-11 by Thinkersbluff