CR6Community/CR-6-touchscreen #82: Possible to have the file name to scoll in the file name windows. ?

dsagdfgsFD, 2022-05-18:


Sebazzz, 2022-05-18:
Yes - I've attempted this, however this was not feasible due to the buggy behavior of the scrolling widget in DWIN.

Thinkersbluff, 2022-05-30:
> Yes - I've attempted this, however this was not feasible due to the buggy behavior of the scrolling widget in DWIN.

FWIW, @Sebazzz  - I modified a version of 61F_RC_260522_v1 to replace the regular text display fields with scrolling text display fields on the filebrowser screen (#67).  Now that we have the new DGUS tool and we now know that the scrolling text window must have a font spacing parameter value of "1", we might be able to make this enhancement work.
The current variable address space does not, however, support filenames longer than 16 characters, and I expect the motherboard firmware handler would need to be modified, to make use of the scrolling text fields.
If you would like to try to implement this again, please advise what variable pointers to put in & I will update the DWIN_SET in this branch of my repo, to support testing the mod:

https://github.com/Thinkersbluff/CR-6-touchscreen_Thinkersbluff_Fork/tree/AddScrollingFileNames

apassiou, 2023-05-24:
Doesnt have to scroll, it can display first X charactes for 5 seconds, then swap to next set etc... should be easier.

closed 2024-12-29 by Thinkersbluff