CR6Community/CR-6-touchscreen #83: How do I use the LEVELING SETTING menu in the Calibrate menu?

junonara, 2022-05-26:
I know that the CR6Community's firmware has a "Fade height" setting function in the "LEVELING SETTINGS" menu. But when I click the "LEVELING SETTINGS" menu in the "Calibrate -> BED LEVELING" menu, nothing happens. If I click LEVELING SETTINGS, the sub menu should be displayed, but the sub menu is not displayed and stays in the LEVELING menu.

The picture of the Fade Height setting in the Community Firmware release 6.1 release notes is as follows.
[image]

I don't know if this is a 3d printer firmware issue or a touch screen firmware issue. Is there another way to enter the LEVELIN SETTINGS menu?

Thinkersbluff, 2022-05-26:
Might be specific to your system.

No problem getting here on my system:

[image]

Which touchscreen firmware do you have loaded?

junonara, 2022-05-26:
I used 61F_RC_090522_v2. And 61F_RC_290422_v1 is the same. The 3D printer firmware is Community 6.1 version, and only Configuration.h has been modified using the configs of cr6-se-v4.5.3-mb.

#define FIL_RUNOUT_ENABLED_DEFAULT false
#define DEFAULT_LEVELING_FADE_HEIGHT 10.0

Thinkersbluff, 2022-05-26:
> I used 61F_RC_090522_v2. And 61F_RC_290422_v1 is the same

Ah, sorry, I should have asked that question first.
You may have found an error in the refactoring. It is 3 am as I write this reply and I have a 2 hr print job running.  I will check tomorrow and get back to you.

Thinkersbluff, 2022-05-26:
So, I definitely see issues on the page itself, which I will address with a new RC, but I had no trouble getting here with v2.  Have not yet tried v1.

[image]

Thinkersbluff, 2022-05-26:
Please try flashing this repaired RC and let me know if it works?
[DWIN_SET.zip](https://github.com/CR6Community/CR-6-touchscreen/files/8781281/DWIN_SET.zip)

junonara, 2022-05-27:
I flashed the DWIN_SET you sent to the touch screen and modified the community firmware source to confirm that the screen output is normal. The community firmware source modification is Marlin\src\lcd\extui\dgus_creality\creality_touch\DGUSDisplayDef.cpp.

[image]

The UI processing part of LEVELING SETTINGS was blocked by an unused definition called HAS_BED_PROBE_SETTINGS. After changing the HAS_BED_PROBE_SETTINGS to HAS_PROBE_SETTINGS and updating the 3D printer, it was possible to enter the LEVELING SETTINGS menu and the menu icons are displayed normally.

[image]

Thanks for your help every time.

Thinkersbluff, 2022-05-27:
> The UI processing part of LEVELING SETTINGS was blocked by an unused definition called HAS_BED_PROBE_SETTINGS. After changing the HAS_BED_PROBE_SETTINGS to HAS_PROBE_SETTINGS and updating the 3D printer, it was possible to enter the LEVELING SETTINGS menu and the menu icons are displayed normally.

I am very happy for you that your edits solved your problem.
I am equally mystified as to why you found that necessary, as my own system still has the original "HAS_BED_PROBE..." line in Marlin\src\lcd\extui\dgus_creality\creality_touch\DGUSDisplayDef.cpp, and it is working fine.

No one should have to edit the CF6.1-Final mainboard firmware, to use the refactored DWIN firmware.

closed 2022-05-27 by Thinkersbluff