CR6Community/CR-6-touchscreen #91: Question: Laser engraving with PWM possible?

Schnema1, 2022-08-14:
Hi dear Team, 

I did follow the tutorial and flashed the LCD and the controller firmware. Now I tried to add a Sculpfun S9 laser module. In order to get enought power, I did connect the 12V with an sperate 12V power supply. The PWM signal I do take from the part fan. In order to get the 12V I did add two 2k resistor in series and picked the 12V to supply the pwm pin of the laser. 

I did connect it to lightburn and tested the pwm function through the console section and the M106 command. It worked well. But once connected to the laser, it will never turn off the laser. If I change the M106 P0 Sxxx to 010 or higher, the laser is blinking. 

Questin:
1: Did aleady someone hooked up a PWM laser module to this software?
2: Is there a "laser mode" in marlin coding?
2.1: If so, how can I enable it?

Thanks for helping me out.

Marc

Sebazzz, 2022-08-24:
Did you connect the grounds of the power supply together? If not, that can cause all kinds of issues.

Schnema1, 2022-08-29:
Thank you.
Yeah, this weekend I have tried to connect both ground together. It does change the beam, but it does not switch it off. Still the laser is burning the wood. I have decided to keep using the printer as printer and start an engraver project with the MKS32 hardware.

Thinkersbluff, 2023-04-20:
This issue seems to be abandoned.

closed 2023-04-20 by Thinkersbluff