CR6Community/CR-6-touchscreen #92: Where is the latest released firmware and DWIN-SET matched files for uploading and flashing??

oliverhbailey, 2022-09-08:
Hello,
This project is the most disorganized github project I have ever seen.  I cannot do a search for a set of working firmware and screen files for the CR6-SE that can be located and useful.  Why can't there be a single place where the CR6 community files and binaries reside that are easy to understand, download, and contain a simple readme file, firmware file, and the DWIN binary to make this easy.  And where are the latest 6.1 binaries? You people should be ashamed because all you do is give open source software a bad name. Think!  You need a readme file, a firmware file, and a DWIN folder file.  There is so much misinformation that clogs up the internet on this topic it is terrible. Hundreds of videos that are useless.
I written nine books in 50 years and some written 20 years ago are still as useful today as back then.  And I cannot locate a set of binaries to flash a CR-6 SE that has a problem and the error message is neither documented or useful.

Thinkersbluff, 2022-09-08:
Hello. Oliver.  Perhaps if you start from The Marlin Repository Releases page, you will find it easier to navigate to the binaries you seek.  https://github.com/CR6Community/Marlin/releases/tag/v2.0.8.1-cr6-community-release-6.1

As stated on the main page of the CR6Community Marlin GitHub project, "This Marlin fork has the goal of cleaning-up the source code changes for the CR-6 so it can be merged upstream."  That goal is now nearing completion, largely through the herculean ongoing efforts of Thinkyhead, Sebazzz and InsanityAutomation - none of whom is being paid for their efforts.  

The CR6Firmware is being held at CF6.1 to provide a stable version from which that upstream effort can be completed.  We do not control the pace or timing of that work.

You may find this upstream PR to be of interest: https://github.com/MarlinFirmware/Marlin/pull/19958

In the meantime, we do our best to keep our users informed as to the location and use of the latest binaries, through 
 - Readme files posted on both the Touchscreen and Marlin repositories, and bundled with each formal release.
 - Our CR6Community Support channel on YouTube [https://www.youtube.com/channel/UCVUcNC8Aa_u6x7ENBKHruvA)
 - I also recently secured a Community License for a GitBook, from the generous folks at GitHub, hoping to provide a consolidated portal to our various disjointed references.  I would welcome your feedback on that, also, while it is still under development: https://app.gitbook.com/invite/7wPsSgkWztlnLzNsSxtp/edPWpzLBtXhcdIE6WkaR

Up until December 2021, the Marlin repository was "the one true source" for all CR6Commaunity Firmware releases. The Discord channel and its genenous volunteer membership our primary support portal. 
In December or January, Creality silently began selling CR6 printers with a more recent version of DGU2 installed, which was not compatible with the DWIN firmware bundled with the motherboard firmware on the Community GitHub.  
We dealt with that situation by spawning a refactored firmware, which is "DGUS-version agnostic" (at least for now, since we cannot stop Creality from doing something like that again, some day.)

Until the upstream Marlin work is finished, I do not know whether the CR6 Community version of the firmware will remain necessary, or whether the upstream Marlin will bundle both motherboard and display firmware in the upstream Marlin 2.1.x.  
Since I keep expecting to see that PR closed, soon, I have not pressed to have a new release of 6.1.x to bundle the refactored display firmware with the motherboard firmware.  

If you think that would be helpful, please let us know.

oliverhbailey, 2022-09-08:
> MarlinFirmware/Marlin#19958

So let me give you a tip.  If you have used download the SD formatter
program from the SD association AND use an 8 Gig SD or MicroSD card,  the
read problems will go away every time.  The problem is a timing problem
with this version of firmware.  Its on the edge of a timing boundary, most
likely due to the scheduler and formatting the card using the SD formatter
program will fix the problem.

Thank you for your reply and have a wonderful day.

Regards,
Oliver Bailey

closed 2022-09-08 by Thinkersbluff

KoenVanduffel, 2022-09-12:
Looks like this [oliverhbailey](https://github.com/oliverhbailey) finds a lot of "This project is the most disorganized github project I have ever seen." projects. I got an almost copy paste of this sentence on my Klipper repo: "This is the most disorganized and poorly documented project I have ever seen." 

Seems this person has a bit of a personal problem!