CR6Community/CR-6-touchscreen #94: Are this JPEG artifacts on the screen?

zappanaut, 2022-12-15:
Hi there!

Thanks a lot for keeping the CR6 alive!

When reviving mine today after months of collecting dust I first flashed some new firmware. I noticed that the screen looked a bit blurry and thought that these were just my fingerprints, but it turned out it looks very much like jpeg artifacts.

Is this due to a glitch during the build process or something unavoidable?

[image]

Thinkersbluff, 2022-12-15:
Yep, probably jpeg artifacts.

Thinkersbluff, 2023-03-13:
The Refactored version 1.1 released 13 March 2023 has been built from scratch, using the original CF6.1 screen bitmaps and minimizing jpg compression.  The colour scheme has also been restored to black/grey/red.

Hopefully, this solves the compression artifacts problem flagged here.

closed 2023-03-13 by Thinkersbluff