CR6Community/CR-6-touchscreen #96: [info] Compatible with CR10 Smart Pro?

boromyr, 2023-02-05:
I wanted to know if this firmware would be compatible with CR10 Smart Pro, obviously with the screen in portrait mode. Thank you.

Any response or opinion releases you from any liability for any damage or malfunction.

boromyr, 2023-02-06:
I read up on the type of screen, and on the CR10 Smart it has already been documented loading your firmware and it seems to work, the screen on the CR10 Smart Pro is almost identical, except for the serial number, some capacitors and some other electronic details. I suspect it's some limited run or a different vendor, but it's still a DGUS2 T5L. A user told me that he used the CR10 Smart screen in a CR10 Smart Pro and also updated the firmware, so I think he can run your CR6 firmware.

Thinkersbluff, 2023-04-20:
This issue seems to be answered.

closed 2023-04-20 by Thinkersbluff