CR6Community/CR-6-touchscreen #98: Stuck on UI loading screen

wheetboi, 2023-03-30:
[image]

Hello, I'm currently stuck on this loading screen and cannot get past it. 

I had V4.2.0 so I followed the README and ended up here.

> 1. Ignore the Touchscreen firmware folder in the CF6.1-Final release zip file.
> 
> 2. INSTEAD, copy the DWIN_SET folder and contents from this release to the micro-SD card you use to flash the display 
> 
> 3. DO NOT copy ANY DGUS2 kernel upgrade files to the DWIN_SET folder before flashing it to your display.

grobux, 2023-03-30:
Hi !
Have you also updated the firmware of the motherboard with the corresponding version?

wheetboi, 2023-03-30:
Hello,

I have not, the guide I was reading said to update that only after I handle the UI screen first.

grobux, 2023-03-30:
According to your screenshot, it looks like the update of the screen is OK.
After rereading the documentation, it seems to me you can keep the last Community Firmware for the motherboard, version 6.1.
Sorry if I'm misleading you, I didn't test it and I didn't flash my printer since quite a long time.
Which version of the motherboard firmware are you using?

wheetboi, 2023-03-30:
I believe the update is OK it's just not going past that full loading bar. 

This is my first time flashing the motherboard. I just tried to flash it with:

[CF6.1-Final-cr6-se-v4.5.3-mb-2021-06-04-16-53.zip](https://github.com/CR6Community/Marlin/releases/download/v2.0.8.1-cr6-community-release-6.1/CF6.1-Final-cr6-se-v4.5.3-mb-2021-06-04-16-53.zip)

However it's still just stuck at that loading bar, nothing else since then.
[CF6.1-Final-cr6-se-v4.5.3-mb-2021-06-04-16-53.zip](https://github.com/CR6Community/Marlin/releases/download/v2.0.8.1-cr6-community-release-6.1/CF6.1-Final-cr6-se-v4.5.3-mb-2021-06-04-16-53.zip)

grobux, 2023-03-30:
And you have a motherboard revision 4.5.3 right?

wheetboi, 2023-03-30:
I have a v1.1.0.3-ERA motherboard, which I believe according to the FAQs means I use the v4.5.3 file 
[image]

grobux, 2023-03-30:
Yes, perfect.
Are you flashing from the stock firmware to the community one, or are you trying to upgrade from another community firmware version?
If you're upgrading from the stock one, you way want to try the previous non-refactored version (only the touchscreen files inside the release ZIP):
https://github.com/CR6Community/Marlin/releases/v2.0.7.3-cr6-community-release-6
https://github.com/CR6Community/Marlin/releases/download/v2.0.7.3-cr6-community-release-6/CR6Comm-CF6-Final-cr6-se-v4.5.3-mb-2021-03-27-15-53.zip

wheetboi, 2023-03-30:
Yes from stock firmware (I believe? I have done zero flashing before today so I assume I'm on everything stock 😂)

I'll give these a shot next. Do I need to do any reversion process or just flash em with the files above

grobux, 2023-03-30:
You can directly flash over the current firmware.

Sorry but I've posted an outdated version in my previous message.
In the ZIP you mentioned before (`CF6.1-Final-cr6-se-v4.5.3-mb-2021-06-04-16-53.zip`), you can find the screen firmware in this path: `/Firmware/Touch screen firmware/CR-6-touchscreen-2021-06-04.zip`

Good luck !

wheetboi, 2023-03-30:
[image]

Well it's definitely the non-refactored version. Still frozen on loading bar with or without trying to flash the mobo

wheetboi, 2023-03-30:
In an attempt to just revert back and start fresh, I flashed back down to the stock firmware on the website and now everything is in Chinese and the touchscreen doesn't work 🙃 

https://www.creality.com/pages/download-cr-6-se?spm=..page_1934481.products_display_1.1&spm_prev=..index.header_1.1

Thinkersbluff, 2023-03-31:
The motherboard and display firmwares are a matched set. There is no benefit to flashing random pairs. 

The interface will stick on the flash screen, if it does not hear from the motherboard. Eventually, you should see a message about the bootloader being confused, which is a reference to the motherboard bootloader. 

You can use M115 to find out which version of firmware is currently flashed to the motherboard. 
If the screen still sticks on the splash screen when M115 confirms that CF6.1 Final is flashed to the motherboard, then I would inspect the serial cable connecting the display to the motherboard, for signs of damage or loose pins. (The serial interface is on the middle two wires of the cable. )

Thinkersbluff, 2023-03-31:
> I flashed back down to the stock firmware on the website and now everything is in Chinese and the touchscreen doesn't work

It is possible that the Creality firmware you flashed requires you to install DGUS2 v4.5 for the touchscreen to work. If you flashed backwards from 4.5 to 3.5 while installing one of the many firmwares you tried, that could stop the touchscreen from working. 

The file and instructions for flashing back to 4.5 are in the Refactored DWIN_SET zip file.

ritchiedc, 2023-04-02:
This is a bit confusing between here and discord (he didn't know about discord originally). He has flashed DGUS2 v4.5 but I don't know if he has tried going back to stock since then.

Thinkersbluff, 2023-04-02:
I am still waiting to hear what M115 tells us about which firmware is on the mainboard.

Thinkersbluff, 2023-04-02:
Unfortunately for others who may encounter this problem, the Author of this Issue solved the problem by trading the printer back to MicroCenter for something else...
I will close this issue.  

If someone else encounters this problem, please try the above recommendations before opening a new report.

closed 2023-04-02 by Thinkersbluff