CR6Community/CR-6-touchscreen #100: usage of dwinos_page_change.bin

Das-Weeman, 2023-05-15:
Hi, can someone please explain whats the usage of this file? i inadvertently flashed only this file and now the lcd is stuck with only a loading bar thats not loading and it wont accept ANY new files that im trying to reflash, resultin in zeros and noughts...

Thinkersbluff, 2023-05-16:
That particular file is automatically generated and inserted into DWIN_SET by the DGUSTools application.  I have no idea what it does.  

This is the first time I have heard of this particular problem.
Thank you for flagging it.  Sorry for your troubles.

The display bootloader should be configured (by the T5L…cfg file) to look for the DWIN_SET folder. It will then only try to flash whatever it finds in that folder, from the few file types that it “recognizes”. 

Can you flash the display with the T5L…cfg file alone?
If not, please try a different card.

Thinkersbluff, 2023-05-16:
_“When i first checked the Dgus2, i was running v4.3 as stock, should i go to
v3.5 with 6.1 bundled lcd fw or v4.5 with the refactored fw for the lcd?“_

I would not upgrade from 4.3 to 4.5.  You would also have to recalibrate the touchscreen to get it working with 4.5, and the Refactored UI runs fine on 4.3.

Das-Weeman, 2023-07-01:
Hi there,

Just a update, i did manage to get everything up and running, CF 6.1 and
all, the main problem that i had was that my sd card adapter developed a
bad connection on it and decided to quit. So i got a new adapter card with
8gb sd card, and boom ..... No hassle installing both main board and lcd
firmware .. but now i have a mesh that looks like the attached pic. I did
the calibration with no bowden installed, so theres no strain on the gauge,
but i still get this mesh... Any advice??

Thank you again for all your effort with the firmware, much appreciated...

Yours truly

Thinkersbluff, 2023-07-01:
I have closed the original issue, since it turned out to be a card failure, not a bug in the firmware. 

I recommend you post an image of your bed mesh on the #bed-leveling channel on the Discord.

closed 2023-07-01 by Thinkersbluff

Das-Weeman, 2023-07-01:
Thank you, I will do that.