CR6Community/CR-6-touchscreen #103: CFG file never updates

HVR88, 2023-12-24:
This is on a 2020 machine.  Everything except CFG has updated, including screen firmware 3.5.

When booting the screen never shows a Creality logo, only two progress bars, one on top and one on bottom of screen. Attempting to flash mainboard to 6.1 then produces an error saying to flash TFT.

Updating to Creality's own 2021 release for TFT and mainboard works every time to put everything back in working order, but it doesn't matter how many hundreds of times 6.1 and this TFT update is attempted, nothing ever ends in a working config.

8GB FAT32 card, formatted in Windows, 4096 b/sect

ritchiedc, 2023-12-25:
I have several questions to start

1. What motherboard do you have?
2. What version of DGUS2 OS was on the display before you first tried flashing it?
3. What zip file(s) did you use for the motherboard firmware and screen firmware?
4. Have you tried flashing more than one version of the community firmware for the screen?

HVR88, 2023-12-25:
Thanks for the reply.

> * What motherboard do you have?
4.5.2

> > * What version of DGUS2 OS was on the display before you first tried flashing it?

1.04 - and I subsequently successfully flashed to 3.5 - everything seems to flash except CFG

> * What zip file(s) did you use for the motherboard firmware and screen firmware?

MB Only: CF6.1-Final-cr6-se-v4.5.2-mb-2021-06-04-16-52.zip  
Screen: Re-Factored.DWIN_SET.Version1.1.2.zip (this is where screen 3.5 came from)

> * Have you tried flashing more than one version of the community firmware for the screen?

No, this is the only version I've ever tried.

Right now running Creality's official "CR-6SE 32bit mainboard+screen firmware new 20211112 28 Jan. 2021"

ritchiedc, 2023-12-25:
Try a different micro sd card to flash the screen. Both the 4.5.2 motherboard and screen can be picky about the card. I had one that worked and then didn't and I had to use a different card.

When you flash the screen record the DGUS2 OS version shown on the top line.

HVR88, 2023-12-26:
I've now tried with 2 new cards. A 2GB Lexar and a 4GB Sandisk.  Same results, regardless of how many times I power cycle to flash.  Looks just like below

Thinkersbluff, 2023-12-27:
Is the T5L…CFG file from the Refactored DWIN_SET  in your DWIN_SET folder?
Can you flash just that file?

HVR88, 2023-12-27:
> Is the T5L…CFG file from the Refactored DWIN_SET in your DWIN_SET folder?

Yes. But it doesn't matter which archive is used nor which cfg - that file never flashes. Not even when I restored/updated to the official Creality release.

> Can you flash just that file?

No difference, CFG never flashes.

Thinkersbluff, 2023-12-27:
If you can not flash the CFG file, that is why your screen does not display correctly.  The CF6.1 loads to a different memory location from the Creality firmware. The CFG file sets that location.
Others have successfully flashed just that file, so the problem is not with the file.
No idea what would cause the screen bootloader routine to selectively disregard that one file type.

Thinkersbluff, 2024-01-04:
If you are still having thi problem, I recommend you delete the current partition volume on the SD card and start again. Create one new volume 8gb MBR primary partition on the card. Format that partition as FAT32, 4096 allocation unit sectors. Then create one folder named DWIN_SET and copy only the T5LCFG file into that folder and flash just that file to your display.

HVR88, 2024-01-11:
@Thinkersbluff first, thank you for the last bit of nudging to get me to try this again.

Against what I thought was my better judgement, I decided to try this one more time, even though I had already done exactly what was suggested multiple times. Wanting to change at least one variable, I decided to try a brand new 8GB card which I'd obtained since last posting.

And... un-be-effing-lievable. This time the CFG actually flashed. Then after that I loaded the full DWIN_SET folder and every file that was supposed to flash did so. Reboot with the mainboard firmware and Community Firmware title/load sequence showed up correctly and finished the process.  Wow.

All I can say is that if someone else finds themselves in this situation, stop repeating the process on the current SD card and go try another one. If that fails a couple of times, try a different card.

Thanks again!

closed 2024-01-11 by HVR88