CR6Community/CR-6-touchscreen #104: Support Touchscreen T5L1 DGUS2 v4.3 2021.10.21.

adelapava, 2024-01-19:
Hello my display has T5L1 DGUS2 v4.3 2021.10.21. and by mistake I flashed with the T5L_UI_DGUS2_V45_20220105............. now what to do? the screen is black nothing even try to flashing with the 3.5 is working what should I do?

Thinkersbluff, 2024-01-20:
What firmware do you have on the display?

adelapava, 2024-01-20:
Before load the kernel  the screen said T5L1 DGUS2 v4.3 2021.10.21
[image]

After applying the kernel appears in red the 
T5L_UI_DGUS2_V45_20220105.bin actually I did remember exactly the red but I just take one as put in thinking was the firmware not the kernel till I saw the end and the red letters

Thinkersbluff, 2024-01-20:
I don't know what firmware loads 21 ICL files.  The Community Firmware loads only 6.

What firmware are you trying to flash to your display,in that photo?

adelapava, 2024-01-20:
The Creality files standard for the ender 3 DWINEnder3 ProMarlin2.0.6.1HW4.2.2

Thinkersbluff, 2024-01-30:
But this repo is specifically for the CR6 printers.  Why flash it to an Ender 3?

adelapava, 2024-01-31:
Well because the screen that Creality sales is the same touchscreen for CR6 then I flash the ender with firmware for the motherboard for ender and the screen have their own firmware but again I use the kernel 4.5 for the touchscreen appear red letters then I realize use wrong SD card inclusive there is a GitHub for the touchscreen of the CR6 and the ender 3 I try with their .bin and still don't have the screen back

https://github.com/InsanityAutomation/Marlin/tree/F1rstLayer_Touchscreens

Thinkersbluff, 2024-01-31:
If you are having problems flashing the Insanity Automation firmware to your Ender3, you need to ask them for help. The CR6 display firmware and Ender3 motherboard firmware will not work with each other.  The display firmware and motherboard firmware are a matched pair.  They only work with each other.

If your display screen went blank when you flashed DGUS2 v4.5, you probably flashed an older version of the display firmware.  Flashing it back to DGUS2 v3.5 should restore the display.

adelapava, 2024-03-23:
Hello I try this but unfortunately not working I found this procedure that I think will help to recover my screen #6 DWIN Display Recovery -Burning Kernel files with PGT05youtu.beBest regards Andres De la pava Sent from my iPhone

closed 2024-12-29 by Thinkersbluff