CR6Community/CR-6-touchscreen #107: CR-6 SE Firmware upgrade not working

famato1953-dev, 2025-09-24:
Hi
 my CR-6 SE 3D Printer failed the upgard, the touch screen does not respond at all when touching any respective tabs.

before i upgraded both firmware and touch screen i took pictures of my system.

my motherboards is CR-ERA v1.1.0.3
my touch screen about tab shows Version 2.0.1.8
and when i started the printer with a blank folder named DWIN_SET it shows T5L1 DUGS2 V4.1 2021.04.14

[image]

[image]

[image]

now the problem i have by finding the firmware for both it's still not working correctly, it shows DGUS V3.5 NOT V4.1
which does not match with my original one, sometime the print homes in at the front left insted of the middle.

can you please assist and email me both firmware that will work with my printer
famato1953@gmail.com i would be greatly appriciated id you can help me, by emailing me directly.

thanking you oncw again in advance.
regards Frank Amato

ritchiedc, 2025-09-24:
First, have you asked for help on the community discord server? You'll likely get better help from the community there.

Second, can you provide the names of the zip files where you got the motherboard and screen firmware?

famato1953-dev, 2025-09-24:
0169cd61ccd233c7ba9c501dec15228b was one and the other one i've accidently deleted itwas on the same page from the roll back part

famato1953-dev, 2025-09-24:
[SR-6 SE Firmware and Touchscreen.zip](https://github.com/user-attachments/files/22505363/SR-6.SE.Firmware.and.Touchscreen.zip) i renamed it as it was too big when unziping it.

[ab244e3ca6d45078d3f4d64c029ef898.zip](https://github.com/user-attachments/files/22505377/ab244e3ca6d45078d3f4d64c029ef898.zip)

famato1953-dev, 2025-09-24:
sorry that was the 2 i used, i did have a third one but still no good

ritchiedc, 2025-09-24:
That is not a zip file name for any release. The screen firmware will be in the following file on this github channel "Re-Factored.DWIN_SET.Version1.1.2.zip

The motherboard firmware will be in "CF6.1.1-Final-cr6-se-v4.5.3+v1.1.0.3ERA-mb-2024-12-16.zip"

Are these the zip files you downloaded?

famato1953-dev, 2025-09-24:
yes i tried them but the touch screen does not work, so i found them on your website.
also tried CF6.1 and that also failed

famato1953-dev, 2025-09-24:
also i noticed it install DGUS2 v3.5 not v4.1?

famato1953-dev, 2025-09-24:
it was V4.1 as per the picture you see all the 000 in all rows.
wen it install the nudates it was v3.5 with the 001, 001, 003 etc

ritchiedc, 2025-09-24:
If you didn't do anything to explicitly add the 3.5 files to dwin then you likely tried flashing the files from version 6. 

The touch screen calibration changed between 3.5 and 4.1 which makes them incompatible and touch will not work when going from one to the other. If you started with 4.1 and it's now 3.5 you will need to flash it back to 4.1. Everything you need is in the Reloaded zip file. 

I'll look up more info. It's difficult doing this on my phone with the GitHub app. That's why I asked if you had posted this problem on the discord server. It's easier to go back and forth getting info to include.

famato1953-dev, 2025-09-24:
thanking you greatly appriciated for your assistance.

ritchiedc, 2025-09-24:
I gave you the wrong zip for the motherboard. I'll go get the correct one

ritchiedc, 2025-09-24:
The correct file for the motherboard is

https://github.com/CR6Community/Marlin/releases/download/v2.0.8.1-cr6-community-release-6.1/CF6.1.1-Final-cr6-se-v4.5.3+v1.1.0.3ERA-mb-2024-12-16.zip

ritchiedc, 2025-09-24:
For the motherboard you simply copy "firmware-20210604-165337.bin" from the "firmware\motherboard firmware" folder to the SD card. The motherboard is expecting the file to be in the root directory of the SD card. You don't need any other files or folders for the motherboard

Thinkersbluff, 2025-09-24:
If you did copy the v45 DGUS2 firmware to your display, that is why the screen is frozen.
Flash that back to v35.
The v45 firmware will only work if you also recalibrate the touchscreen.

Thinkersbluff, 2025-09-24:
There is too much info in this video, but it does show you how to calibrate the screen and how to switch between v35 and v45.

https://youtu.be/0xwFGiyg4Z8?feature=shared

famato1953-dev, 2025-09-24:
How do I get the 3.5 again. I copied your file is thought it was p art of it

famato1953-dev, 2025-09-24:
> Flash the V35 file from this folder:
>
> https://github.com/CR6Community/CR-6-touchscreen/tree/extui/src/DGUS2%20kernel%20files
>
> The video walks you through the process.
>
>
> Get Outlook for iOS<https://aka.ms/o0ukef>

i am very grateful  of all your assistance in resolving my problem.
Great it's working now.

regards Frank

ritchiedc, 2025-09-24:
I'm glad you got it working. It is likely you will want additional help as you begin using your printer. You can join the discord by going to the following link. 

https://discord.gg/KKhjQsjq

BadCreation, 2026-02-21:
> The correct file for the motherboard is
> 
> https://github.com/CR6Community/Marlin/releases/download/v2.0.8.1-cr6-community-release-6.1/CF6.1.1-Final-cr6-se-v4.5.3+v1.1.0.3ERA-mb-2024-12-16.z

Do you happen to have a firmware for the motherboard v4.5.2?